thliebig/openEMS #36: h5readatt_octave.cc:3:10: fatal error: hdf5.h: No such file or directory

darkstar007, 2018-04-25:
Hi,

I'm getting the following error when I try the Patch_Antenna_Array tutorial file:
'''
----------------------------------------------------------------------
 | openEMS 64bit -- version v0.0.35-28-g6133dea
 | (C) 2010-2016 Thorsten Liebig <thorsten.liebig@gmx.de>  GPL license
 ----------------------------------------------------------------------
        Used external libraries:
                CSXCAD -- Version: v0.6.2-80-g6ab8a3a
                hdf5   -- Version: 1.10.0
                          compiled against: HDF5 library version: 1.10.0-patch1
                tinyxml -- compiled against: 2.6.2
                fparser
                boost  -- compiled against: 1_62
                vtk -- Version: 6.3.0
                       compiled against: 6.3.0

--------

RunFDTD: Warning: Max. number of timesteps was reached before the end-criteria of -50dB was reached...
        You may want to choose a higher number of max. timesteps...
Time for 30000 iterations with 588744.00 cells : 306.45 sec
Speed: 57.63 MCells/s
 ----------------------------------------------------------------------
 | nf2ff, near-field to far-field transformation for openEMS
 | (C) 2012-2014 Thorsten Liebig <thorsten.liebig@gmx.de>  GPL license
 ----------------------------------------------------------------------
calculating far field at phi=[0 90] deg...
warning: function "h5readatt_octave" not found, trying to run "setup"
warning: called from
    ReadHDF5Attribute at line 15 column 9
    ReadNF2FF at line 22 column 12
    CalcNF2FF at line 140 column 7
    Patch_Antenna_Array at line 228 column 7
setting up openEMS matlab/octave interface
compiling oct files
h5readatt_octave.cc:3:10: fatal error: hdf5.h: No such file or directory
 #include "hdf5.h"
          ^~~~~~~~
compilation terminated.
HDF5 library path found at: /usr/lib/x86_64-linux-gnu/hdf5/openmpi/libhdf5.so
/usr/lib/x86_64-linux-gnu/hdf5/serial
HDF5 include path found at: /usr/include/hdf5/openmpi/hdf5.h
/usr/include/hdf5/serial
warning: mkoctfile: building exited with failure status
error: 'h5readatt_octave' undefined near line 22 column 19
error: called from
    ReadHDF5Attribute at line 22 column 10
    ReadNF2FF at line 22 column 12
    CalcNF2FF at line 140 column 7
    Patch_Antenna_Array at line 228 column 7
'''

Which seems to be telling me it cannot find hdf5.h, and yet it then says it found it in the openmpi and serial directories.

Any suggestions to how to proceed? I'm assuming that this is happening in the nf2ff octave interface - how can I replicate this without re-running the whole simulation?

Thanks again,

Matt

tmolteno, 2018-06-03:
The following changes to the setup.m file (you have to modify the file in the installed location will fix this problem.) Notice the pipe to the head function, if you had openmpi installed then the find command returned two lines and the location of the header file got confused.

[res, fn_so] = unix('find /usr/lib     -name libhdf5.so');
[res, fn_h]  = unix('find /usr/include -name hdf5.h | head -1');

[hdf5lib_dir, hdf5lib_fn, ext] = fileparts(fn_so);
disp(["HDF5 library path found at: " hdf5lib_dir])

[hdf5inc_dir, hdf5inc_fn, ext] = fileparts(fn_h);
disp(["HDF5 include path found at: " hdf5inc_dir])
mkoctfile("h5readatt_octave.cc", ["-L" hdf5lib_dir], ["-I" hdf5inc_dir], "-L hdf5")

thliebig, 2018-06-03:
Sounds about right. You prepare a commit and pull request? ;)

thliebig, 2018-11-17:
I guess this can be closed then?

closed 2018-11-17 by thliebig

rmccomiskie, 2021-02-01:
Hello. I just installed the latest openEMS & Octave following the instructions here: 
https://openems.de/index.php/Compile_from_Source.html
for Ubuntu 20.04 LTS.

After adding paths to Octave, I ran setup.m as instructed.

> octave:7> setup
> setting up openEMS matlab/octave interface
> compiling oct files
> HDF5 library path found at: /usr/lib/x86_64-linux-gnu/hdf5/serial/libhdf5.so
> /usr/lib/x86_64-linux-gnu/hdf5/openmpi
> HDF5 include path found at: /usr/include/hdf5/serial/hdf5.h
> /usr/include/hdf5/openmpi
> h5readatt_octave.cc:3:10: fatal error: hdf5.h: No such file or directory
>     3 | #include "hdf5.h"
>       |          ^~~~~~~~
> compilation terminated.
> warning: mkoctfile: building exited with failure status

I found a fix here: https://github.com/thliebig/openEMS/issues/36,  tmolteno commented on Jun 3, 2018

However, the changes were not included in my setup.m.
Did they really get merged in 2018?

thliebig, 2021-02-01:
Yes they were. Did you install the latest version?

arisnik, 2021-11-16:
Hello, i have the same problem in Ubuntu 20.04 LTS and i have the latest version. Any ideas?

Scowt, 2021-12-05:
I had a similar issue on Ubuntu 21.10 - solved by adding a second '| head'

[res, fn_so] = unix('find /usr/lib -name libhdf5.so');
to
[res, fn_so] = unix('find /usr/lib -name libhdf5.so | head -1');