thliebig/openEMS #38: munmap_chunk(): invalid pointer when running openEMS on Arch Linux

Keziolio, 2018-08-31:
I followed the first tutorial but I can't get openEMS to run properly, it crashes with the error in the title. (or free(): invalid pointer)

I tried 3 times to get it to generate a debug build by hacking your updater thing but for some reason it doesn't work, di you have an official method to add debug symbols so this bug report is somewhat useful?

```> gdb --args ./openEMS /tmp/tmp.xml 
GNU gdb (GDB) 8.1.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Registered pretty printers for UE4 classes
Reading symbols from ./openEMS...(no debugging symbols found)...done.
(gdb) r
Starting program: /tmp/openems/bin/openEMS /tmp/tmp.xml
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
 ---------------------------------------------------------------------- 
 | openEMS 64bit -- version v0.0.35
 | (C) 2010-2016 Thorsten Liebig <thorsten.liebig@gmx.de>  GPL license
 ---------------------------------------------------------------------- 
        Used external libraries:
                CSXCAD -- Version: v0.6.2
                hdf5   -- Version: 1.10.2
                          compiled against: HDF5 library version: 1.10.2
                tinyxml -- compiled against: 2.6.2
                fparser
                boost  -- compiled against: 1_67
                vtk -- Version: 6.3.0
                       compiled against: 6.3.0

munmap_chunk(): invalid pointer

Program received signal SIGABRT, Aborted.
0x00007ffff5e6fd7f in raise () from /usr/lib/libc.so.6
(gdb) 

```

thliebig, 2018-09-01:
I think I had a similar issue on my Manjaro (Arch based). But I just recompiled everything and it went away. But I will have another look into it...

thliebig, 2018-09-01:
I just checked I did make some changes to CSXCAD and openEMS, please make sure you have the latest version. From your message above it shows you only have the latest **release** version...