thliebig/openEMS #39: Lorentz / Drude materials are causing a segfault in openEMS v0.0.35

tvelliott, 2018-09-12:
I recently upgraded to v0.0.35 and started getting segfault for the dispersive material types (lorentz/drude).  I'll attach a sample file from the examples.
[tutorial_m.txt](https://github.com/thliebig/openEMS/files/2376853/tutorial_m.txt)

tvelliott, 2018-09-13:
Some more info...  running the dispersive example:
octave share/openEMS/matlab/examples/other/Metamaterial_PlaneWave_Drude.m
Segfault is right after it calculates and outputs the smallest time step information..
```

~$ octave share/openEMS/matlab/examples/other/Metamaterial_PlaneWave_Drude.m

warning: Syntax for InitFDTD has changed, use: "InitFDTD('NrTS', 50000, 'EndCriteria', 1e-05)" instead! Legacy support enabled.
warning: called from
    InitFDTD at line 48 column 5
    share/openEMS/matlab/examples/other/Metamaterial_PlaneWave_Drude.m at line 55 column 10
args = "MTM_PW_Drude.xml" -vvv
 ---------------------------------------------------------------------- 
 | openEMS 64bit -- version v0.0.35-37-gcc5a709
 | (C) 2010-2018 Thorsten Liebig <thorsten.liebig@gmx.de>  GPL license
 ---------------------------------------------------------------------- 
	Used external libraries:
		CSXCAD -- Version: v0.6.2-84-g196a6ce
		hdf5   -- Version: 1.10.0
		          compiled against: HDF5 library version: 1.10.0-patch1
		tinyxml -- compiled against: 2.6.2
		fparser
		boost  -- compiled against: 1_62
		vtk -- Version: 6.3.0
		       compiled against: 6.3.0

openEMS - verbose level 3
Read openEMS xml file: MTM_PW_Drude.xml ...
Read openEMS Settings...
Read Geometry...
-----------------------------------------
 Property #0 Type: "Excitation" Name: "excite"
 Primitive Count 	: 1
 Coordinate System 	: 0
  -- Primitives: --
  Primitive #0 Type: "Box" Priority: 0
  Primary Coord-System: 2 Mesh Coord-System: 0 Bound-Box Coord-System: 0
  Bounding Box (Valid: 1): P1: (-5,-0.75,-249) P2: (5,0.75,-249)
  Transform: None
  Start: -5,-0.75,-249
  Stop : 5,0.75,-249
 --- Excitation Properties --- 
  Type: 0
  Active directions: 1,1,1
  Excitation	: 0, 1, 0
  Weighting	: 1, 1, 1
  Propagation Dir: 0, 0, 0
  Delay		: 0
-----------------------------------------
 Property #1 Type: "LorentzMaterial" Name: "drude"
 Primitive Count 	: 1
 Coordinate System 	: 0
  -- Primitives: --
  Primitive #1 Type: "Box" Priority: 10
  Primary Coord-System: 2 Mesh Coord-System: 0 Bound-Box Coord-System: 0
  Bounding Box (Valid: 1): P1: (-5,-0.75,-125) P2: (5,0.75,125)
  Transform: None
  Start: -5,-0.75,-125
  Stop : 5,0.75,125
 --- LorentzMaterial --- 
  Isotropy	: 1
  Epsilon_R	: 1, 1, 1
  Kappa		: 0, 0, 0
  Mue_R		: 1, 1, 1
  Sigma		: 0, 0, 0
  Density	: 0
  Lorentz model order:	1
 Epsilon Plasma Frequency #0:	5e+09,5e+09,5e+09
 Mue Plasma Frequency #0:	5e+09,5e+09,5e+09
 Epsilon Lorentz Pole Frequency #0:	0,0,0
[... 46 lines cut ...]
	libvtkIOXMLParser-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkIOXMLParser-6.3.so.6.3 (0x00007efe39dc5000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007efe39b9b000)
	libvtkIOLegacy-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkIOLegacy-6.3.so.6.3 (0x00007efe398f5000)
	libvtkIOPLY-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkIOPLY-6.3.so.6.3 (0x00007efe396da000)
	libvtkIOGeometry-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkIOGeometry-6.3.so.6.3 (0x00007efe393bf000)
	libvtkIOCore-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkIOCore-6.3.so.6.3 (0x00007efe3914a000)
	libvtkCommonExecutionModel-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkCommonExecutionModel-6.3.so.6.3 (0x00007efe38ea0000)
	libvtkCommonDataModel-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkCommonDataModel-6.3.so.6.3 (0x00007efe3893f000)
	libvtkCommonMisc-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkCommonMisc-6.3.so.6.3 (0x00007efe38729000)
	libvtkCommonSystem-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkCommonSystem-6.3.so.6.3 (0x00007efe38515000)
	libvtksys-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtksys-6.3.so.6.3 (0x00007efe382ce000)
	libvtkCommonTransforms-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkCommonTransforms-6.3.so.6.3 (0x00007efe3809f000)
	libvtkCommonMath-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkCommonMath-6.3.so.6.3 (0x00007efe37e7f000)
	libvtkCommonCore-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkCommonCore-6.3.so.6.3 (0x00007efe3795c000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007efe375da000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007efe373c3000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007efe37024000)
	libCGAL.so.12 => /usr/lib/x86_64-linux-gnu/libCGAL.so.12 (0x00007efe36e01000)
	/lib64/ld-linux-x86-64.so.2 (0x00007efe3ceb6000)
	libaec.so.0 => /usr/lib/x86_64-linux-gnu/libaec.so.0 (0x00007efe36bf9000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007efe369f1000)

```

```

tvelliott, 2018-09-13:
Sorry for multiple comments on this, but I did manage to track down the issue to this:

in the file:  FDTD/extensions/operator_ext_lorentzmaterial.cpp

This call in two places:
```

CSProperties* prop = m_Op->GetGeometryCSX()->GetPropertyByCoordPriority(coord, vPrims, true);
			if ((mat = prop->ToLorentzMaterial())) { }
```

prop returns as NULL if the the material is Lorentz/Debye material type.  For now, I'm just hard coding the lorentz parameters if prop is NULL and it appears to be working ok.  

Maybe because I'm using newer libboost?  I'll add another comment if I figure out why the NULL is getting returned.

thliebig, 2018-11-17:
I applied your patch. Thanks for that and sorry that it took me so long...

closed 2018-11-17 by thliebig