thliebig/openEMS #41: openems.de webpage is down

michal777, 2018-10-31:
I can't access openems webpage. What's wrong? Will it come back soon?

thliebig, 2018-10-31:
The host has major issues. I don't know why. All attempts to fix it over the last couple of weeks did not help. Plans to move the page are under way, but I don't know how long this will take.

brnold, 2018-11-07:
Do you have the tutorials hosted/available somewhere else?

thliebig, 2018-11-07:
The tutorials are all inside the binary and source packages and on github of course...

odo2063, 2018-11-10:
Wieviel Webspace bräuchtest du denn? 

How much Webspace would you need?

thliebig, 2018-11-17:
I think this can be closed now. The new site is online. We just need to think about a new/better documentation. The old wiki will remain a static page now...

closed 2018-11-17 by thliebig