thliebig/openEMS #46: Some spurious signal appears at the end of the simulation and increases in energy

rubund, 2019-02-15:
Hi

I could not find the openEMS forum (which I remember I have seen earlier), so I am asking here.

I am simulating a microstrip connected to a coaxial cable. I am making a 2d field dump on top of board. Most of the simulation looks great, but near the end when it seems like the excitation has died out, some spurious signal suddenly appears at the connection between the microstrip and the coax. It builds up in energy and eventually fills the entire grid with "high" energy. I was wondering if you know why this happens, and how it can be prevented.

The code is attached below:

```
#!/usr/bin/env octave

pkg load openems
pkg load csxcad

close all
clear
clc

physical_constants;
unit = 1e-6; % specify everything in um
MSL_length = 50000;
MSL_width = 600;
substrate_thickness = 254;
substrate_epr = 3.66;
stub_length = 12e3;
f_max = 7e9;

FDTD = InitFDTD();
FDTD = SetGaussExcite( FDTD, f_max/2, f_max/2 );
BC   = {'PML_8' 'PML_8' 'MUR' 'MUR' 'MUR' 'MUR'};
FDTD = SetBoundaryCond( FDTD, BC );

CSX = InitCSX();
resolution = c0/(f_max*sqrt(substrate_epr))/unit /50; % resolution of lambda/50
mesh.x = SmoothMeshLines( [9000 11000] , resolution/4, 1.5 ,0 );
mesh.x = SmoothMeshLines( [-MSL_length 8000 mesh.x MSL_length], resolution, 1.5 ,0 );
mesh.y = SmoothMeshLines( [0:50:2300], resolution, 1.3, 0);
mesh.y = SmoothMeshLines( [-15*MSL_width -mesh.y mesh.y 15*MSL_width] , resolution, 1.3, 0);
mesh.z = SmoothMeshLines( [-10*substrate_thickness [-2300:50:-254] -200 -130 -70 0 50 130 200 [254:50:2300] 10*substrate_thickness], resolution/4 );
CSX = DefineRectGrid( CSX, unit, mesh );

CSX = AddMaterial( CSX, 'RO4350B' );
CSX = SetMaterialProperty( CSX, 'RO4350B', 'Epsilon', substrate_epr );
start = [mesh.x(1),   mesh.y(1),  0];
stop  = [10000, mesh.y(end), substrate_thickness];
CSX = AddBox( CSX, 'RO4350B', 0, start, stop );

CSX = AddMetal( CSX, 'PEC' );

%%% ground plane
start = [mesh.x(1),   mesh.y(1),   0];
stop  = [10000, mesh.y(end), 0];
CSX = AddBox( CSX, 'PEC', 999, start, stop );

%% MSL port
portstart = [ mesh.x(1), -MSL_width/2, substrate_thickness];
portstop  = [ 0,  MSL_width/2, 0];
[CSX,port{1}] = AddMSLPort( CSX, 999, 1, 'PEC', portstart, portstop, 0, [0 0 -1], 'ExcitePort', true, 'FeedShift', 10*resolution, 'MeasPlaneShift',  MSL_length/3);

portstart = [mesh.x(end), 0, substrate_thickness+180];
portstop  = [10000          ,  0, substrate_thickness+180];

[CSX,port{2}] = AddCoaxialPort( CSX, 999, 2, 'PEC', '', portstart, portstop, 'x', 200, 1500, 1700, 'MeasPlaneShift', MSL_length/3, 'FeedShift', 10*resolution);

start = [0,  -MSL_width/2, substrate_thickness];
stop  = [10000,  MSL_width/2, substrate_thickness];
CSX = AddBox( CSX, 'PEC', 999, start, stop );

%% Field dump
CSX = AddDump(CSX,'Et','DumpMode', 2);
start = [mesh.x(1)   mesh.x(1)   substrate_thickness];
stop  = [mesh.x(end) mesh.x(end) substrate_thickness];
CSX = AddBox(CSX,'Et',0 , start,stop);

 
%% write/show/run the openEMS compatible xml-file
Sim_Path = 'tmp';
Sim_CSX = 'msl.xml';

[status, message, messageid] = rmdir( Sim_Path, 's' ); % clear previous directory
[status, message, messageid] = mkdir( Sim_Path ); % create empty simulation folder

WriteOpenEMS( [Sim_Path '/' Sim_CSX], FDTD, CSX );
CSXGeomPlot( [Sim_Path '/' Sim_CSX] );
RunOpenEMS( Sim_Path, Sim_CSX );
```

Thank you very much in advance.

Best regards
Ruben

thliebig, 2019-02-15:
This is indeed not the right place for such questions, please use the [forum](http://openems.de/forum/) and attach the m-file

closed 2019-02-15 by thliebig

rubund, 2019-02-16:
> please use the [forum](http://openems.de/forum/) and attach the m-file

Thanks. http://openems.de is dead, and it leads to me to a https page. And when adding forum it ended up on a https page where the forum is not. Also the link from the openEMS wiki page did not work yesterday. Google was not much help either. But your direct link works.