thliebig/openEMS #47: API version issue with HDF5

matthuszagh, 2019-02-18:
I'm seeing an API version issue with HDF5. For instance if I run the rectangular waveguide tutorial I see this output:

```
octave> source rect_waveguide.m 
args = "rect_wg.xml" 
 ---------------------------------------------------------------------- 
 | openEMS 64bit -- version v0.0.35-45-gde23172
 | (C) 2010-2018 Thorsten Liebig <thorsten.liebig@gmx.de>  GPL license
 ---------------------------------------------------------------------- 
	Used external libraries:
		CSXCAD -- Version: v0.6.2-85-g55899d0
		hdf5   -- Version: 1.10.4
		          compiled against: HDF5 library version: 1.10.4
		tinyxml -- compiled against: 2.6.2
		fparser
		boost  -- compiled against: 1_69
		vtk -- Version: 8.1.2
		       compiled against: 8.1.2

Create FDTD operator (compressed SSE + multi-threading)
FDTD simulation size: 27x12x122 --> 39528 FDTD cells 
FDTD timestep is: 7.79642e-13 s; Nyquist rate: 24 timesteps @2.67217e+10 Hz
Excitation signal length is: 1225 timesteps (9.55062e-10s)
Max. number of timesteps: 10000 ( --> 8.16327 * Excitation signal length)
Create FDTD engine (compressed SSE + multi-threading)
Running FDTD engine... this may take a while... grab a cup of coffee?!?
[@        4s] Timestep:         3496 || Speed:   34.5 MC/s (1.144e-03 s/TS) || Energy: ~1.09e-30 (-81.26dB)
Time for 3496 iterations with 39528.00 cells : 4.00 sec
Speed: 34.54 MCells/s 
warning: legend: 'best' not yet implemented for location specifier, using 'northeast' instead
warning: legend: 'best' not yet implemented for location specifier, using 'northeast' instead
error: API version api-v52+ found in .oct file function 'h5readatt_octave'
       does not match the running Octave (API version api-v53)
       this can lead to incorrect results or other failures
       you can fix this problem by recompiling this .oct file
error: called from
    ReadHDF5Attribute at line 22 column 10
    ReadHDF5FieldData>ReadHDF5FieldData_octave at line 97 column 34
    ReadHDF5FieldData at line 29 column 19
    PlotHDF5FieldData at line 25 column 8
    rect_waveguide at line 121 column 1
```

I believe I have the latest master versions of both Octave and OpenEMS. Have you encountered this issue? I'm using Arch Linux.

thliebig, 2019-02-18:
I have not seen this before. have you tried to rebuild the .oct file? E.g. by just running "setup" (in Octave with openEMS in path) again?

matthuszagh, 2019-02-18:
That works! thanks for the quick reply.

closed 2019-07-10 by matthuszagh