thliebig/openEMS #54: Compile on Ubuntu 16.04 Fails

st68, 2019-08-09:
Attempting to compile OpenEMS on Ubuntu 16.04.10

cloned from github on 2019-08-08

Issue: AppCSXCAD doesn't compile
Suspect wrong or missing version of qt...

From the log file:
/usr/bin/ld: cannot find -lQt5::Core
/usr/bin/ld: cannot find -lQt5::Widgets
/usr/bin/ld: cannot find -lQt5::Xml

Needed to add these packages to get this far -- might be good to add to install instructions for 16.04
    tcl-vtk
    python-vtk
    libvtk-java

Tried adding to the following to AppCSXCAD/CMakeLists.txt (both NEW and OLD):

if(POLICY CMP0028)
  cmake_policy(SET CMP0028 NEW)
endif()

which changed error messages, but to no avail.

#
# HERE FOLLOWS LOTS OF DATA THAT MIGHT BE RELEVANT
#
# ------------------- SYSTEM --------

Ubuntu 16.04.10

from /proc/version
  Linux version 4.4.0-157-generic (buildd@lgw01-amd64-037)
  (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.10) )
  #185-Ubuntu SMP Tue Jul 23 09:17:01 UTC 2019

# ------ LOADED PACKAGES (output below) ---------

sudo apt-get install \
	build-essential \
	cmake \
	git \
	libhdf5-dev \
	libvtk5-dev \
	libboost-all-dev \
	libcgal-dev \
	libtinyxml-dev \
	libqt4-dev \
	libvtk5-qt4-dev \
	octave \
	liboctave-dev \
	gengetopt \
	help2man \
	groff \
	pod2pdf \
	bison \
	flex \
	libhpdf-dev \
	libtool \
	libopenmpi-dev \
	paraview \
	octave \
	libcgal-qt5-dev \
	tcl-vtk \
	python-vtk \
	libvtk-java \

# this fails
./update_openEMS.sh ~/opt/openEMS

# this works
./update_openEMS.sh ~/opt/openEMS -disable-GUI

# this fails
./update_openEMS.sh ~/opt/openEMS --with-hyp2mat --with-CTB --with-MPI

# this works
./update_openEMS.sh ~/opt/openEMS --with-hyp2mat --with-CTB --with-MPI --disable-GUI

# ------------ FAILURE LOG RESULTS -- MANY VARIANTS OF THESE -----------------

CMake Warning (dev) at CMakeLists.txt:134 (ADD_EXECUTABLE):
  Policy CMP0028 is not set: Double colon in target name means ALIAS or
  IMPORTED target.  Run "cmake --help-policy CMP0028" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  Target "AppCSXCAD" links to target "Qt5::Xml" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?
This warning is for project developers.  Use -Wno-dev to suppress it.

# ------------ FAILURE LOG RESULTS -- NEAR END OF LOG -----------------
-- Generating done
-- Build files have been written to: /tmp/tmp.mi1NWEQCsE/AppCSXCAD-prefix/src/AppCSXCAD-build
[ 75%] Performing build step for 'AppCSXCAD'
Scanning dependencies of target AppCSXCAD_automoc
[ 20%] Automatic moc for target AppCSXCAD
Generating moc_AppCSXCAD.cpp
[ 20%] Built target AppCSXCAD_automoc
Scanning dependencies of target AppCSXCAD
[ 40%] Building CXX object CMakeFiles/AppCSXCAD.dir/main.cpp.o
[ 60%] Building CXX object CMakeFiles/AppCSXCAD.dir/AppCSXCAD.cpp.o
[ 80%] Building CXX object CMakeFiles/AppCSXCAD.dir/AppCSXCAD_automoc.cpp.o
[100%] Linking CXX executable AppCSXCAD
/usr/bin/ld: cannot find -lQt5::Core
/usr/bin/ld: cannot find -lQt5::Widgets
/usr/bin/ld: cannot find -lQt5::Xml
collect2: error: ld returned 1 exit status
CMakeFiles/AppCSXCAD.dir/build.make:192: recipe for target 'AppCSXCAD' failed
... more fail, fail, fail

# ------------------ OUTPUT OF apt-get ABOVE (for version numbers) ---------------------
# --- note:
# ---   cmake 3.5.1
# ---   qt4-dev 4.4.8.7

Reading package lists... Done
Building dependency tree       
Reading state information... Done
bison is already the newest version (2:3.0.4.dfsg-1).
build-essential is already the newest version (12.1ubuntu2).
flex is already the newest version (2.6.0-11).
libqt4-dev is already the newest version (4:4.8.7+dfsg-5ubuntu2).
libtinyxml-dev is already the newest version (2.6.2-3).
libtool is already the newest version (2.4.6-0.1).
gengetopt is already the newest version (2.22.6+dfsg0-1).
groff is already the newest version (1.22.3-7).
help2man is already the newest version (1.47.3).
libboost-all-dev is already the newest version (1.58.0.1ubuntu1).
libcgal-dev is already the newest version (4.7-4).
libcgal-qt5-dev is already the newest version (4.7-4).
libhpdf-dev is already the newest version (2.2.1-1.1).
libopenmpi-dev is already the newest version (1.10.2-8ubuntu1).
libvtk-java is already the newest version (5.10.1+dfsg-2.1build1).
libvtk5-dev is already the newest version (5.10.1+dfsg-2.1build1).
libvtk5-qt4-dev is already the newest version (5.10.1+dfsg-2.1build1).
paraview is already the newest version (5.0.1+dfsg1-4).
pod2pdf is already the newest version (0.42-5).
python-vtk is already the newest version (5.10.1+dfsg-2.1build1).
tcl-vtk is already the newest version (5.10.1+dfsg-2.1build1).
cmake is already the newest version (3.5.1-1ubuntu3).
git is already the newest version (1:2.7.4-0ubuntu1.6).
libhdf5-dev is already the newest version (1.8.16+docs-4ubuntu1.1).
liboctave-dev is already the newest version (4.0.0-3ubuntu9.2).
octave is already the newest version (4.0.0-3ubuntu9.2).

Liath, 2020-04-13:
FWIW, I'm compiling on Ubuntu@18.04 with the following Dockerfile:
```Dockerfile
FROM ubuntu:18.04

RUN echo "APT::Acquire::Retries \"3\";" > /etc/apt/apt.conf.d/80-retries

# Saves us from hanging on tzdata update later
RUN ln -fs /usr/share/zoneinfo/UTC /etc/localtime
# Install build tools and libraries
RUN apt update -y && apt install -y autoconf bison build-essential cmake       \
  fig2dev flex gengetopt git libcgal-dev libhdf5-dev libhpdf-dev liboctave-dev \
  libtinyxml-dev libvtk6-dev libvtk6-qt-dev octave

WORKDIR /tmp/
RUN git clone --recursive https://github.com/thliebig/openEMS-Project.git

WORKDIR /tmp/openEMS-Project
# mkoctfile per http://openems.de/forum/viewtopic.php?t=808#p3578
RUN ./update_openEMS.sh /opt/openEMS --with-hyp2mat --with-CTB --with-MPI &&   \
  cd /opt/openEMS/share/openEMS/matlab && mkoctfile -I/usr/include/hdf5/serial \
  -L/usr/lib/x86_64-linux-gnu/hdf5/serial h5readatt_octave.cc
COPY .octaverc /root/.octaverc
# RUN rm -rf /tmp/openEMS-Project

WORKDIR /tmp/run
COPY simulation.m /tmp/run/simulation.m
RUN octave ./simulation.m
```

thliebig, 2021-04-11:
The original issue is rather old. Are there still open issues for Ubuntu 18.04 and 20.04?

omriseemann, 2021-08-15:
I have issue building on ubuntu 21, I get this error:

/usr/bin/ld: /home/omri/Programs/anaconda3/lib/libQt5Xml.so.5.9.7: undefined reference to `QXmlUtils::isPublicID(QString const&)@Qt_5_PRIVATE_API'
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/10/../../../x86_64-linux-gnu/libSM.so: undefined reference to `uuid_generate@UUID_1.0'
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/10/../../../x86_64-linux-gnu/libSM.so: undefined reference to `uuid_unparse_lower@UUID_1.0'
collect2: error: ld returned 1 exit status
make[5]: *** [CMakeFiles/AppCSXCAD.dir/build.make:331: AppCSXCAD] Error 1
make[4]: *** [CMakeFiles/Makefile2:96: CMakeFiles/AppCSXCAD.dir/all] Error 2
make[3]: *** [Makefile:149: all] Error 2
make[2]: *** [CMakeFiles/AppCSXCAD.dir/build.make:131: AppCSXCAD-prefix/src/AppCSXCAD-stamp/AppCSXCAD-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:103: CMakeFiles/AppCSXCAD.dir/all] Error 2
make: *** [Makefile:103: all] Error 2

thliebig, 2022-12-31:
It seems cmake has found the wrong Qt?

thliebig, 2022-12-31:
I close this old issue. If build issues still exist, please reopen

closed 2022-12-31 by thliebig