thliebig/openEMS #55: SIGSEGV for infinite dipole example

lemoer, 2019-08-25:
I just installed openEMS on Archlinux and tried to simulate using MATLAB. I tried to simulate the examples [infDipol](https://github.com/thliebig/openEMS/blob/master/matlab/examples/antennas/infDipol.m) and [Bent_Patch_Antenna](https://github.com/thliebig/openEMS/blob/master/matlab/Tutorials/Bent_Patch_Antenna.m) and openEMS crashed with SIGSEGV.

I am using:
```
 ---------------------------------------------------------------------- 
 | openEMS 64bit -- version v0.0.35
 | (C) 2010-2016 Thorsten Liebig <thorsten.liebig@gmx.de>  GPL license
 ---------------------------------------------------------------------- 
	Used external libraries:
		CSXCAD -- Version: v0.6.2
		hdf5   -- Version: 1.10.5
		          compiled against: HDF5 library version: 1.10.5
		tinyxml -- compiled against: 2.6.2
		fparser
		boost  -- compiled against: 1_69
		vtk -- Version: 8.2.0
		       compiled against: 8.2.0
```
and matlab R2019a.

I used valgrind to get a little bit more information:
```
==1546== Memcheck, a memory error detector
==1546== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==1546== Using Valgrind-3.14.0 and LibVEX; rerun with -h for copyright info
==1546== Command: openEMS tmp.xml
==1546== 
 ---------------------------------------------------------------------- 
 | openEMS 64bit -- version v0.0.35
 | (C) 2010-2016 Thorsten Liebig <thorsten.liebig@gmx.de>  GPL license
 ---------------------------------------------------------------------- 
	Used external libraries:
		CSXCAD -- Version: v0.6.2
		hdf5   -- Version: 1.10.5
		          compiled against: HDF5 library version: 1.10.5
		tinyxml -- compiled against: 2.6.2
		fparser
		boost  -- compiled against: 1_69
		vtk -- Version: 8.2.0
		       compiled against: 8.2.0

==1546== 
==1546== Process terminating with default action of signal 11 (SIGSEGV): dumping core
==1546==  General Protection Fault
==1546==    at 0x4BB2054: uw_update_context_1 (unwind-dw2.c:1395)
==1546==    by 0x4BB2546: uw_init_context_1 (unwind-dw2.c:1607)
==1546==    by 0x4BB2EDA: _Unwind_Resume (unwind.inc:235)
==1546==    by 0x4906054: ??? (in /usr/lib/libopenEMS.so.0.0.35)
==1546==    by 0x1FFEFFFC6F: ???
==1546==    by 0x4: ???
==1546==    by 0x385F4C4D4F: ???
==1546==    by 0x4DBA16E: ParameterScalar::Evaluate() (in /usr/lib/libCSXCAD.so.0.6.2)
==1546==    by 0x3: ???
==1546==    by 0x78616D78: ???
==1546==    by 0x89490EAD3C466FF: ???
==1546==    by 0x1FFEFFFD2F: ???
==1546== 
==1546== HEAP SUMMARY:
==1546==     in use at exit: 443,038 bytes in 4,091 blocks
==1546==   total heap usage: 8,483 allocs, 4,392 frees, 1,062,770 bytes allocated
==1546== 
==1546== LEAK SUMMARY:
==1546==    definitely lost: 112 bytes in 1 blocks
==1546==    indirectly lost: 367 bytes in 4 blocks
==1546==      possibly lost: 0 bytes in 0 blocks
==1546==    still reachable: 442,559 bytes in 4,086 blocks
==1546==         suppressed: 0 bytes in 0 blocks
==1546== Rerun with --leak-check=full to see details of leaked memory
==1546== 
==1546== For counts of detected and suppressed errors, rerun with: -v
==1546== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
fish: “valgrind openEMS tmp.xml” terminated by signal SIGSEGV (Address boundary error)
```

If you need more information, feel free to ask.

thliebig, 2019-08-25:
You are using the release version which has a bug causing this. Please try the latest version from git.
[See here](https://github.com/thliebig/openEMS-Project#clone-build-and-install)

lemoer, 2019-08-26:
Thank you. Works fine.

(I was looking into the known_bugs and known_problems file as well as the github issues but didn't find anything.)

closed 2021-04-11 by thliebig