thliebig/openEMS #64: MPI issue with python interface

matthuszagh, 2019-12-19:
I'm experiencing an openmpi issue when using the python interface. I'm getting the following error message:
```
Create FDTD operator (compressed SSE + multi-threading)
*** The MPI_Comm_size() function was called before MPI_INIT was invoked.
*** This is disallowed by the MPI standard.
*** Your MPI job will now abort.
```
This occurs when attempting to call `FDTD.Run()`. So, for instance, attempting 
```
$ python Simple_Patch_Antenna.py
```
where `Simple_Patch_Antenna.py` is from the openems python tutorials directory. I built openems with MPI support and `mpirun` is in `PATH`.

Have you encountered this? I haven't experienced any issues when using the matlab/octave interface.

thliebig, 2019-12-19:
No, but I never tried openmpi support with the python interface. Try to disable the MPI support it is not really useful in any case ;)

matthuszagh, 2019-12-19:
That worked. I'll close this. Thanks so much for the quick reply @thliebig!

closed 2019-12-19 by matthuszagh