thliebig/openEMS #71: Help to run the examples from the first time.

leoheck, 2020-06-27:
Hi, I am trying to learn this awesome tool. I am starting with the examples on `matlab/examples/antennas`
Then I am seeing this.

```
RunFDTD: Warning: Max. number of timesteps was reached before the end-criteria of -50dB was reached... 
	You may want to choose a higher number of max. timesteps... 
Time for 30000 iterations with 70004.00 cells : 131.53 sec
Speed: 15.97 MCells/s 
calculating far field at phi=[0 90] deg...
 ---------------------------------------------------------------------- 
 | nf2ff, near-field to far-field transformation for openEMS 
 | (C) 2012-2014 Thorsten Liebig <thorsten.liebig@gmx.de>  GPL license
 ---------------------------------------------------------------------- 
warning: function "h5readatt_octave" not found, trying to run "setup"
warning: called from
    ReadHDF5Attribute at line 15 column 9
    ReadNF2FF at line 22 column 12
    CalcNF2FF at line 141 column 7
    Patch_Antenna at line 189 column 7
setting up openEMS matlab/octave interface
compiling oct files
HDF5 library path found at: /usr/lib/x86_64-linux-gnu/hdf5/serial
HDF5 include path found at: /usr/include/hdf5/serial
g++: error: h5readatt_octave.cc: No such file or directory
g++: fatal error: no input files
compilation terminated.
warning: mkoctfile: building exited with failure status
error: 'h5readatt_octave' undefined near line 22 column 19
error: called from
    ReadHDF5Attribute at line 22 column 10
    ReadNF2FF at line 22 column 12
    CalcNF2FF at line 141 column 7
    Patch_Antenna at line 189 column 7
warning: gh_manager::do_close_all_figures: some graphics elements failed to close.
make: *** [Makefile:31: Patch_Antenna] Error 1
```

Do you have any idea how to fix this?
I've installed openEMS from the OS package manager on Ubuntu 20.04.
And I am running the examples from the repo I cloned from the GitHub.

leoheck, 2020-06-27:
I was able to go over it by placing this `h5readatt_octave.cc` in my `/usr/local/share/CSXCAD/matlab/` folder.
The build with the `setup.m` passed but it was not able to run it.

Then I was able to fix it manually with this (this a subset of my Makefile)
Running from here `~/openEMS/matlab/examples/antennas`
```
fix_missing_h5readatt_octave:
	sudo cp ../../h5readatt_octave.cc /usr/share/octave/packages/openems-0.0.35/

INC_PATH = $(dir $(shell find /usr/include -name hdf5.h | head -1))
LIB_PATH = $(dir $(shell find /usr/lib -name libhdf5.so))

build:
	mkoctfile -I$(INC_PATH) -L$(LIB_PATH) -lhdf5 ../../h5readatt_octave.cc
```

thliebig, 2022-12-31:
Is this still open? Please try to build completely from github.

closed 2022-12-31 by thliebig