thliebig/openEMS #72: Packaging Python interface for PyPI

matthuszagh, 2020-08-05:
I believe it would be very convenient for users if the openems and csxcad interfaces were available for installation on [PyPI](https://pypi.org/). The current installation process makes using these interfaces in external libraries difficult. Is this something that might be considered?

thliebig, 2020-08-11:
I agree that this would be very nice indeed, but I'm not sure how difficult that might be at the end.
openEMS and CSXCAD depend on a large number of external C/C++ libraries and I'm unsure how to package that especially since many of this packages have their own python package too (Qt, vtk, hdf5, just to name a few)...
I think to compile and install the python interface on Linux should not be too hard? I surely need to improve the documentation for it.
For Windows my idea would be to ship everything ready to go with a small python setup file to just copy/install it to the right location?
If you have experience in this area, feel free make suggestions and we can discuss it further...

matthuszagh, 2020-08-11:
Let me give this some thought and if I'm able to get something to work I'll post back here with the exact build process and then you can decide if you'd like to use it to upload to pypi.

I agree that the manual installation process isn't too tricky. It imposes the limitation, however, that any python packages that depend on openems must also be installed manually.

As for Windows unfortunately I'm not sure how helpful I'll be as I have almost no experience with it. However, if I am able to get this packaged on linux, I'll test it out on Windows and see if I can get it working there too.

thliebig, 2020-08-11:
Yeah, Windows is the biggest concern, because everything has to be compiled with the fitting MS compiler and this took my a long time to even figure out. Thus getting it done on Linux is one thing, but it will not help you at all for windows...

matthuszagh, 2020-08-21:
After giving this some thought, I think your current solution makes the most sense. Packaging this would require a fair amount of work and require shipping large wheels since openems has some heavy dependencies. Moreover, the current installation method, while manual, is pretty straightforward.

I'm going to close this for now. If someone else has the motivation to attempt this feel free to comment and we can reopen it.

closed 2020-08-21 by matthuszagh