thliebig/openEMS #75: Python/CSXCAD AddBox error

mysol, 2021-07-23:
Hello!
I've got a problem with command CSX.AddBox in python wrapper... Compiler returns an error:
`Traceback (most recent call last):`
  `File "dipole.py", line 48, in <module>`
    `box=CSX.AddBox(substrate, priority=0, start=start, stop=stop )`
`AttributeError: 'CSXCAD.CSXCAD.ContinuousStructure' object has no attribute 'AddBox'`

All examples have the same error with "**AddBox**"

thliebig, 2021-07-23:
The syntax should be:
`box=substrate.AddBox(priority=0, start=start, stop=stop)`

mysol, 2021-07-23:
> The syntax should be:
> `box=substrate.AddBox(priority=0, start=start, stop=stop)`

yes, it works fine, but...
[image]
[image]

```
import os, tempfile
from pylab import *
from mpl_toolkits.mplot3d import Axes3D

from CSXCAD import CSXCAD

from openEMS.openEMS import openEMS
from openEMS.physical_constants import *

Sim_Path = os.path.join(tempfile.gettempdir(), 'dipole')

unit = 1e-3 # all length in mm

f_max = 0.5e9
lambda0 = C0/f_max
step = lambda0/50

SimBox = np.array([200, 200, 150])

CSX = CSXCAD.ContinuousStructure()
mesh = CSX.GetGrid()
mesh.SetDeltaUnit(unit)

mesh.AddLine('x', linspace(-SimBox[0]/2, SimBox[0]/2, num=SimBox[0]))
mesh.AddLine('y', linspace(-SimBox[1]/2, SimBox[1]/2, num=SimBox[1]))
mesh.AddLine('z', linspace(-SimBox[2]/3, SimBox[2]*2/3, num=SimBox[2]))

f0 = 400e6
fc = 0.5*f0

patch_width  = 32 #
# patch length in y-direction
patch_length = 40
substrate_thickness = 1.524

patch = CSX.AddMetal( 'patch' ) # create a perfect electric conductor (PEC)
start = [-patch_width/2, -patch_length/2, substrate_thickness]
stop  = [ patch_width/2 , patch_length/2, substrate_thickness]
patch.AddBox(priority=10, start=start, stop=stop) # add a box-primitive to the metal property 'patch'

CSX_file = os.path.join(Sim_Path, 'simp_patch.xml')
if not os.path.exists(Sim_Path):
    os.mkdir(Sim_Path)
CSX.Write2XML(CSX_file)
os.system(r'AppCSXCAD "{}"'.format(CSX_file))
```

closed 2021-07-23 by mysol