thliebig/openEMS #84: how to calculate time interval ?

fwt1888, 2022-01-30:
Hi, I came cross a problem. I want to know where time interval (i.e. dt) is defined or calculated in your code, because I think it is needed for calculation of f0 and fc.

thliebig, 2022-01-31:
Well the calculation of dt is complicated (operator.cpp CalcTimestep). After the simulation it can be determined/read from several places.
Maybe you can say what you want to do?? 
And would have the forum been a better place for this question?

fwt1888, 2022-02-02:
I want to set Gaussian excitation, but I'm not sure about the calculation of f0 and fc. I think timestep is needed in the forum related. Could you tell me the forum you use to calculate timestep? Thanks.

MrHighVoltage, 2022-03-29:
AFAIK the maximum timestep is limited by the Grid settings (the minimum distance between two adjacent points). The timestep is, from my experience, much smaller then 1/(2*(fc+f0)), which is needed to keep within Nyquist–Shannon sampling theorem.
Apart from that, I don't think you really need to know the time interval for the calculation of f0 and fc, because OpenEMS does that by itself. You only have to check if the time-step (as a result of your grid-resolution) somehow matches your required simulation frequencies.

closed 2022-08-28 by fwt1888