thliebig/openEMS #88: How to generate the exe file ? (run in windows)

WEIXUEP, 2022-06-25:
Hi, I successfully run the source code in ubuntu and generate the file in floder 'opt' by using the update_openEMS.sh file,  but I don't know how to generate the exe file that can run in the windows.

thliebig, 2022-06-26:
The windows build is a very long and complicated one. Please just download the prebuild binaries [openEMS.de](https://openems.de/download/win64/)

closed 2022-06-26 by thliebig

biergaizi, 2023-05-11:
An easy route is MinGW / MSYS2. If there's a need for testing the latest development code, one can use MinGW to build openEMS on Windows in a way that is almost identical to Unix. Performance should be similar to MSVC, since MinGW uses the same GCC to generate x86_64 code as  a native PE binary, it's not an emulation.

I also managed to cross-compile openEMS to Windows from ArchLinux via MinGW, then openEMS can be run under Wine (ArchLinux was used because it contains many user packages for MinGW cross-compile, such as Boost or VTK). It means that one can debug and test openEMS completely on Linux without ever touching Windows or any Microsoft development tool, very useful for development.

I'll soon document the procedures at GitHub Discussion.