thliebig/openEMS #89: libopenEMS.so.0.0.35 doesn't have a SONAME

yurivict, 2022-06-26:
```
====> Running Q/A tests (stage-qa)
Warning: /disk-samsung/freebsd-ports/science/openems/work/stage/usr/local/lib/libopenEMS.so.0.0.35 doesn't have a SONAME.
Warning: pkg(8) will not register it as being provided by the port.
Warning: If another port depend on it, pkg will not be able to know where it comes from.
Warning: It is directly in /usr/local/lib, it is probably used by other ports.
```

thliebig, 2022-12-31:
I'm not sure what this means or how you got here...

yurivict, 2022-12-31:
SONAME options need to be added to the link command for the shared library.

thliebig, 2023-01-01:
Well I'm still not really sure what to do about it or why I should care?
Feel free to make a pull request or recommend a cmake instruction to fix this?

yurivict, 2023-01-01:
One way is:

```
set(PROJECT_SOVERSION 0) # bump if ABI breaks

set_target_properties(openEMS PROPERTIES VERSION
    ${PROJECT_SOVERSION}.${PROJECT_VERSION_MINOR}.${PROJECT_VERSION_PATCH})
  set_target_properties(openEMS PROPERTIES SOVERSION ${PROJECT_SOVERSION})
```

This is similar to: https://github.com/baresip/baresip/pull/2187/files

thliebig, 2023-01-02:
I just checked. This is already in for a long time, which leaves me even more puzzled about your error message?

https://github.com/thliebig/openEMS/blob/master/CMakeLists.txt#L213

biergaizi, 2023-01-18:
I can confirm that a SONAME indeed exists on my FreeBSD server.

    [worker@freebsd ~/opt/openems/lib]$ objdump -p libopenEMS.so.0.0.35 | grep SONAME
      SONAME               libopenEMS.so.0

The version in FreeBSD port is likely very outdated, which is not uncommon if a package only has little attention. I think this bug can be closed. Yet another lesson for us to always consider testing the latest development version before filling any bug report...

yurivict, 2023-01-18:
The latest commit doesn't have this problem.

closed 2023-01-18 by yurivict