thliebig/openEMS #92: Cant install the octave integration: error: 'isOctave' undefined near line 17, column 4

fcolecumberri, 2022-07-10:
I am running on linux, I have already added the:

```octave
addpath("/usr/local/share/openEMS/matlab/");
```

to the `.octave` file and then execute the `setup` instruction, I only get the next error:

```txt
error: 'isOctave' undefined near line 17, column 4
error: called from
    setup at line 17 column 4
```

Any help?

ra1nb0w, 2022-11-14:
same issue; have you found a fix? thanks

fcolecumberri, 2022-11-14:
> same issue; have you found a fix? thanks

No

ra1nb0w, 2022-11-15:
resolved running the following command in Octave
```
addpath('/usr/share/CSXCAD/matlab');
addpath('/usr/share/openEMS/matlab');
savepath
```
then you can exit and re-open Octave and `setup` works fine.

closed 2022-12-31 by thliebig