thliebig/openEMS #99: Feature Request: Benchmarking Demos

biergaizi, 2023-01-13:
For openEMS users, it would be insightful to know how different hardware and software configurations (CPU, memory, multi-threading, MPI clustering) affect simulation speed via a set of standard tests. However, the current demos provided in openEMS are meant for teaching purposes, not benchmarks. It would be useful to add some demos with longer runtime (e.g. 5 to 10 minutes) for benchmarking purposes, perhaps by using a smaller mesh than what's strictly necessary. Demos on the use of different boundary conditions (e.g. PEC vs PML) would also be helpful to show their performance impact.

thliebig, 2023-01-14:
Yes I agree. At some point I did have a simple benchmark script with increasing mesh domains and different number of threads. But I cannot find it anymore. 

Feel free to give it a go ;)

biergaizi, 2023-06-13:
If anyone has missed it, I hacked together a simple benchmark suite using pre-existing demos and it has been used for evaluating performance on various systems, as part of my performance optimization attempt. It can be found here:

https://github.com/biergaizi/openems-bench

It's still not a complete benchmark. It does not test with increasing domain sizes, nor does it test every combination of extensions, so sometimes the result can be confusing to interpret.