thliebig/openEMS #102: Feed resistance in AddMSLPort is incorrectly oriented

jmwilson, 2023-01-21:
https://github.com/thliebig/openEMS/blob/0342eefd276336b882057fc87fcc317f9961ccc2/matlab/AddMSLPort.m#L259-L261

This sets the direction of the feed resistance to `idx_height-1`. In the example of a microstrip line with propagation along the x-axis (=0), width measured along the y-axis (=1), and excitation oriented toward the z-axis (=2), `idx_height` is 2, so the resistance is oriented perpendicular to the line. If `idx_height` were 0, then this would pass -1 to `AddLumpedElement` and cause an error at some point.

Changing this to `idx_prop` would create a zero-length lumped element which is not allowed.

thliebig, 2023-01-22:
Did you get into an invalid port? Because otherwise I would just assume that this "-1" is because Matlab counts from 1... and everyone else starts an index at 0. I think this is only this transformation. You find this in a lot of places.

If you check the ["AddLumpedElement"](https://github.com/thliebig/CSXCAD/blob/master/matlab/AddLumpedElement.m) then it expects 0,1,2 as directions, not 1,2,3 what the matlab index would be...

jmwilson, 2023-01-22:
No, I didn't actually run into that error and forgot about Matlab's 1-based indexing. The issue about it picking the direction still seems to be the case.

thliebig, 2023-01-22:
I'm not sure what you mean, the height direction sounds correct and that the index is -1 does too (1/2/3 --> 0/1/2).  Please test the port in different directs to see if it works or not. If it does not, please supply the file such that I can reproduce it and can have a look at it...

thliebig, 2023-02-11:
reopen if you still see the issue

closed 2023-02-11 by thliebig