thliebig/openEMS #113: openEMS.pyx AssertionError due to checking non-canonical path name

biergaizi, 2023-05-03:
Currently, running openEMS's example Python scripts on macOS always fails with the following error:

```
$ python3 MSL_NotchFilter.py 
Traceback (most recent call last):
  File "/Users/gentoo/code/openEMS-Project/openEMS/python/Tutorials/MSL_NotchFilter.py", line 103, in <module>
    FDTD.Run(Sim_Path, cleanup=True)
  File "openEMS/openEMS.pyx", line 489, in openEMS.openEMS.openEMS.Run
AssertionError
```

This is caused by an oversight of an assertion in `openEMS.pyx`:

    os.chdir(sim_path)
    # ...
    assert os.getcwd() == sim_path

The problem here is that `sim_path` is not a canonicalized path name, so the assertion would fail if the path we're switching into contains a symbolic link. This problem affects all operating systems, it's not limited to macOS. But on macOS, the problem is especially serious, since macOS's `/tmp` is a link to `/private/tmp` by default. Thus, it causes an AssertionError in all the included Python examples. 

```
$ python3
Python 3.11.3 (main, May  3 2023, 02:26:34) [GCC 12.2.0] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import os
>>> os.path.realpath("/tmp")
'/private/tmp'
>>> os.chdir("/tmp")
>>> os.getcwd()
'/private/tmp'
```

Instead of doing `assert os.getcwd() == sim_path`, we should do `assert os.getcwd() == os.path.realpath(sim_path)` instead. I'll submit my patch in a Pull Request to fix this problem.

biergaizi, 2023-05-03:
I opened Pull Request #114 to fix this problem. It has been successfully tested on macOS.

thliebig, 2023-05-03:
Running with "python -O" should avoid this with the unpatched version as it ignores asserts

closed 2023-05-03 by thliebig

wisnukusuma, 2025-01-24:
>  "python -O" 

Would it affect to the process solver calculation, and others? Since everytime I need to run every openEMS included need that command. I can't fixed the code manually because I installed with pre-build wheel python the latest one.

biergaizi, 2025-03-16:
> @wisnukusuma: Would it affect to the process solver calculation, and others?

No. The solver is a C++ library called from Python, so it's completely independent.

> Since everytime I need to run every openEMS included I need that command. I can't fixed the code manually because I installed with pre-build wheel python the latest one.

The latest git version has solved this problem. Consider testing it out.