thliebig/openEMS #124: The isOctave() function isn't available in Octave

yurivict, 2023-09-10:
openEMS uses isOctave [here](https://github.com/thliebig/openEMS/blob/master/TESTSUITE/run_testsuite.m#L10) but octave-

info/octave.info-9 provides an example implementation of such function:
```
   Example: 

     %%
     %% Return: true if the environment is Octave.
     %%
     function retval = isOctave
       persistent cacheval;  % speeds up repeated calls

       if isempty (cacheval)
         cacheval = (exist ("OCTAVE_VERSION", "builtin") > 0);
       end

       retval = cacheval;
     end
```

You also define this function in matlab/WriteHDF5.m:
> isOctave = exist('OCTAVE_VERSION','builtin') ~= 0;

thliebig, 2023-09-19:
well this variable "isOctave" was meant to already be available from the context where the test-suite is started.
But this is all very very old code and I'm not sure if it is still in any use. Otherwise feel free to define it properly.
But I don't think that we need anything fancy as the example above as this is nothing that gets called more than once or a few times... A one-liner would be still sufficient I think

thliebig, 2026-09-17:
I think the issue was that the CSXCAD matlab path was not added? The "isOctave" function can be found there. Both the CSXCAD and openEMS matlab path should be added e.g. to the  .octaverc, see [here](https://docs.openems.de/en/latest/octave/install.html#unix-linux)

closed 2026-09-17 by thliebig