thliebig/openEMS #125: Unused primitive

GMellar, 2023-09-17:
I was giving the FreeCAD exporter a shot (https://github.com/LubomirJagos/FreeCAD-OpenEMS-Export) and some issues with metal sheet appeared. The example I attached is a simple microstrip line which is modelled with a metal layer. Then there is a stub as copper sheet. The dimensions of the stub don't make sense at the moment this is just for testing purpose. 

When viewing the design with AppCSXCAD it seems fine but running the simulation gives me the error "Warning: Unused primitive (type: PolyhedronReader) detected in property: CopperSheet!". 

[PCB.zip](https://github.com/thliebig/openEMS/files/12642624/PCB.zip)

LubomirJagos, 2023-09-17:
hi, thanks for giving shot to my GUI addon, seems like there is no gridline in Z coord, you can try make new grid category with 1 line in Z and put copper sheet stub into it and move it in priority list to the top to be sure it overcomes all other grids in Z to be sure there is gridline going through that sheet, then that error message will dissapear.
That message is there due no gridlines are going through that object.

LubomirJagos, 2023-09-17:
could you put this question into my discussion and close this as it's not real issue connected to solver core and it is confusing when this kind of question is opened as issue here, thanks

GMellar, 2023-09-18:
Unfortunately that does not change anything. There is a mesh line in Z direction where the sheet is. Even if I set one extra grid with one line in Z direction there is still the warning.

LubomirJagos, 2023-09-18:
could you post here your freecad file together with settings file saved in .ini? Conducting sheets as material category was one of last things which I added and they are running ok, but your problem is to put gridlines on right place, I'm pretty sure there is missing line in Z but without your freecad and simulation settings file it's hard to figure out how to do it.

GMellar, 2023-09-19:
I attached the freecad and the init file. There is a lane inside and there is a generated mesh line. 
[Test.zip](https://github.com/thliebig/openEMS/files/12656742/Test.zip)

LubomirJagos, 2023-09-19:
Hi, so I briefly experiment with your geometry and found out this is my issue in exporter plugin, the easiest solution for you:
    - extrude stub plane in Z direction as 35um
    - change it's material for PEC or copper or something
    
The thing is I just added new material conducting sheet months ago and never tried any simulation with it, so since I played with your model I see it's seems not to be running with 3D objects and conducting sheet should be 2D plane in some X,Y,Z plane. I'm not sure about this, but seems for me that it's not possible to use it let's say as coax coating.
Need to think about these restriction and add them into GUI addon with proper checkers and generators to generate box with zero thickness.
What I looked for openEMS tutorials there is just MSL_losses using conducting sheet as parameter for microstripline which is possible in my plugin, but haven't find example using it for geometry.
I will think about this, but for your purpose seems to be enough to model that stub as box and define it as metal.

Thanks for your example, I will add this into addon, but first need to think about some checkers and what and how to generate geometry for conducting sheet to make it user friendly, ie. at least display some error messages for user when using conducting sheet for geometry objects to inform user what will happen and for better understanding how it's used.

This will be also good question for Thorsten if conducting sheet can be used just for 2D planar objects or it's possible to use it for 3D objects as coax coating and so.

Let me now if this helped.

thliebig, 2023-09-19:
> This will be also good question for Thorsten if conducting sheet can be used just for 2D planar objects or it's possible to use it for 3D objects as coax coating and so.

Currently the conducting sheet can only be a sheet (as the name suggests). Bulk/volume material should be modeled using "normal" lossy material instead. E.g. just as copper with a sigma of 56e6 S/m
The sheet material is really meant as conductive traces with e.g. 35um thickness and such...

If you want to discuss this here we can keep this open, otherwise just close it

LubomirJagos, 2023-09-20:
So current usage of conductive sheet is for plane like this:
     - cartesian coords: XY, XZ, YZ plane
     - cylindrical coords: XY plane
     
I tried assign loaded plane polyhedron (plane in XY) from STL into conductive sheet and solver was complain that's there is no object even when gridline was crossing it, so conductive sheet is usable just for AddBox() for planes?
Also is there difference due order of rectangle vertices?
    CSX = AddBox(CSX,'CopperSheet',9800,[0 0 zCoord],[-1 10 zCoord]);
    and
    CSX = AddBox(CSX,'CopperSheet',9800,[-1 10 zCoord], [0 0 zCoord]);

Does it change like conductive sheet is taken like it's top or bottom direction in Z? Does it have some effect on results? (due in first I expect sheet is like layer going up to air, in second is going inside substrate)

thliebig, 2023-09-21:
Yes only boxes are currently supported. Maybe polygons too? I would need to check for that. 
But even these objects should be 2D only but maybe it is enough if they are 2D after the grid mapping (again long time ago).
The question about top/bottom I do not understand? But I think both your boxes should give an identical result.
The sheet thickness is only used inside the conductive (loss) model, there is no "real" thickness...

LubomirJagos, 2023-09-29:
I made corrections for conductive sheet and made script generation right, here is release for current corrected version, I removed all releases and left just one which should be correct, need to test it since I was playing with git and I made lot of mess, but decided to left just one version to be sure that everybody who will download it have that right last release version which OK and it's running and problems are easily spotted :D

https://github.com/LubomirJagos/FreeCAD-OpenEMS-Export/releases

thliebig, 2024-04-01:
I think this can be closed then? BTW it would have been a better fit in the openEMS-Project Discussions Section

closed 2024-04-01 by thliebig