thliebig/openEMS #130: Building on Enterprise Linux 8 (RHEL/Alma/Oracle/Rocky)

KJ7LNW, 2023-12-08:
The following patch is required for systems running python 3.6.  (I would set up a pull request, but I am unable to do that at the moment.)

```patch
diff --git a/python/setup.py b/python/setup.py
index 0ac0ba0..b1acb0b 100644
--- a/python/setup.py
+++ b/python/setup.py
@@ -43,5 +43,5 @@ setup(
   url = 'https://openEMS.de',
   packages=["openEMS", ],
   package_data={'openEMS': ['*.pxd']},
-  ext_modules = cythonize(extensions, language_level = "3")
+  ext_modules = cythonize(extensions, language_level = 3)
  )
```

 this prevents errors like the following:

```py
Traceback (most recent call last):
  File "setup.py", line 46, in <module>
    ext_modules = cythonize(extensions, language_level = "3")
  File "/usr/lib64/python3.6/site-packages/Cython/Build/Dependencies.py", line 888, in cythonize
    ctx = c_options.create_context()
  File "/usr/lib64/python3.6/site-packages/Cython/Compiler/Main.py", line 596, in create_context
    self.cplus, self.language_level, options=self)
  File "/usr/lib64/python3.6/site-packages/Cython/Compiler/Main.py", line 94, in __init__
    self.set_language_level(language_level)
  File "/usr/lib64/python3.6/site-packages/Cython/Compiler/Main.py", line 100, in set_language_level
    if level >= 3:
```

## See also:
- https://github.com/thliebig/CSXCAD/issues/37
- https://github.com/thliebig/QCSXCAD/issues/15

thliebig, 2023-12-11:
Thanks for the feedback. I'm a bit puzzled by this problem as this is a typical way to define the language level as far as google searches do indicate.  
But this change seems to be reasonable...

KJ7LNW, 2023-12-12:
Older versions of python wont auto-convert from string to integer, thus removing quotes fixes it.

KJ7LNW, 2023-12-12:
More specifically, I think I read that "older versions of python's `input(?)` function" won't auto-convert from string to integer.  I tried to find the link again but it's not forthcoming.  I'm not sure what `input` function it was referring to, but thats what I remember.  FYI in case its useful.