thliebig/openEMS #135: Python interface build process cannot find necessary headers

aWZHY0yQH81uOYvH, 2023-12-20:
Certain python distributions (namely the one provided by Homebrew on macOS) do not provide the system include paths when building the python bindings, resulting in build failure.

See https://github.com/thliebig/openEMS-Project/discussions/167#discussioncomment-7903125 for more details.

Possible fixes
 * Figure out a cross-platform way of finding the default system include paths and specify them explicitly
 * Modify the compiler flags (in a way supported by both clang and gcc) so that system include paths will not be overwritten
 * Modify the directory structure (or use a symlink) so that header files are present in the same directory as the python input files
 * Modify references to headers to use headers in the repository via a relative path

Also https://github.com/thliebig/CSXCAD/issues/38

thliebig, 2023-12-20:
I'm not sure this sounds like a viable solution. For example do I never use system include paths to build or install openEMS and dependencies.
For example my default build command on Linux (and similar on Windows) looks like this:
` python setup.py build_ext -I/home/<username>/opt/include -L/home/<username>/opt/lib -R/home/<username>/opt/lib`

In other words I always reference the full path to all includes and library locations.
I would recommend doing something similar on macOS?

thliebig, 2023-12-20:
This is btw also what I recommend in the readme inside the python folder

aWZHY0yQH81uOYvH, 2023-12-20:
Adding the python bindings to the Homebrew formula to short circuit this entire problem looks pretty trivial: https://docs.brew.sh/Python-for-Formula-Authors#installing-bindings

thliebig, 2023-12-20:
Personally I would prefer a solution in which the homebrew formula would solve this. This is exactly what the update_openEMS.sh does for building from source on Linux as well...

aWZHY0yQH81uOYvH, 2023-12-21:
I agree. I'll put together a small PR soon to add the python bindings as an option in the Homebrew formula.

I apologize for being a bit naive about focusing on the default include paths since that was the difference between working and not for me.

closed 2024-03-11 by thliebig