thliebig/openEMS #138: Can't install on manjaro linux, can't install on windows

MaximGi, 2024-03-30:
I've been spending most of the last 24 hours trying to install a working version of OpenEMS.

First on manjaro linux, I gave up because of a persisting "No module named Cython" error appearing while trying to install both csxcad and openems python modules in a virtual environment (in which cython is already installed) and after spending a lot of time installing AUR dependencies.

I am now trying to install on windows. I managed to unpack the openEMS install and located in C:\openEMS.
Trying to build and install the python modules (Install guide, Windows, Build Modules From Source) using the command :
`python setup.py build_ext -IC:\openEMS\include -LC:\openEMS`
(Because installing to prebuild .whl didn't worked either since the import fails because the DLL's aren't found, even after defining the OPENEMS_INSTALL_PATH global variable).

It first threw an error because I don't have the MSVC build tools, which wasn't specified. After installing the former, I retried the above command and it fails while building 'CSXCAD.CSProperties' extension by reaching a 100+ count of errors.
The errors (C2039 and C2065) are all a variation of "XXX is not a member of CSPropLumpedelement"

I'll stop trying to use this seemingly very good module because the installation process is way too complicated.

thliebig, 2024-04-01:
I think this should have been opened on the "openEMS-Project" repo?

I guess on Linux you just need to install the missing python libraries like cython. The message was quite helpful?

On Windows you should better not try to compile the extensions yourself but just install the pre-compiled extensions as per instructions in the python folder...

closed 2024-04-01 by thliebig

MaximGi, 2024-04-01:
Hi,

I'll open the windows specific issue on the other repo.

- On linux : I did. As indicated, in my issue ticket : ""No module named Cython" error appearing [...] in a virtual environment (in which cython is already installed)"
- On windows, as indicated : "(Because installing to prebuild .whl didn't worked either since the import fails because the DLL's aren't found, even after defining the OPENEMS_INSTALL_PATH global variable)."