thliebig/openEMS #141: Feature request: progress bar

physicsnerd, 2024-06-11:
When running FDTD.Run, it'd be great if there was some sort of progress bar so I could tell where my simulation was at/whether it was a lost cause.

thliebig, 2024-06-12:
If you have a good idea what that progress bar should track let me know. I'm doing FDTD for 20+ years now and I currently would have no idea ;)

Tracking e.g. steps vs max. number steps is bad because you should always set max number of steps way to large and let the energy decrement decide when to stop. Therefore you never know beforehand at what number of steps a simulation is done.
And tracking the energy decay (decrement) does not really work either, because first the energy rises and then it decays uneven. Most of the time quick at first and (much) slower later...

But as I said, any ideas are welcome...

physicsnerd, 2024-06-17:
I think tracking, for instance, both of these would be reasonable - I'm aware the max steps is not actually when the simulation will be done and the energy decay is not monotonic, but this is more a feature request for a sanity check that, e.g., each step isn't taking way too long, or that the energy decay isn't occurring excessively slowly, or even how close one is to the max number of steps. Maybe this is less a progress bar in the sense of 'towards completion' and more a progress bar in the sense of 'progress is occurring at a reasonable rate' haha.

biergaizi, 2024-08-29:
Also, as we're on it... The current simulation progress shows speed in "seconds per timestep". But this number is usually too small to read. I think it's better to change this to "timesteps per second", which has better readability. If it says "1000 timesteps per second", everyone can do some basic mental math to guess how long it's going to run.