thliebig/openEMS #142: Documentation Request: MultiGrid

ALEEF02, 2024-07-08:
There are functions regarding embedding cylindrical coordinates in a Cartesian system. I believe this will be useful for vias. Some examples can be found in the [openEMS Python definition](https://docs.openems.de/python/openEMS/openEMS.html#openEMS.openEMS.SetMultiGrid) and in the [C source code](https://github.com/thliebig/openEMS/blob/master/FDTD/operator_cylindermultigrid.h). However, there is no example of the implementation of these functions in the [old wiki](https://wiki.openems.de) or the new documentation. Please give some insight on using this feature!

JTSvejda, 2024-07-09:
> There are functions MultiGrid regarding embedding cylindrical coordinates in a Cartesian system. I believe this will be useful for vias. Some examples can be found in the openEMS Python definition <https://docs.openems.de/python/openEMS/openEMS.html#openEMS.openEMS.SetMultiGrid> and in the C source code <https://github.com/thliebig/openEMS/blob/master/FDTD/operator_cylindermultigrid.h>. However, there is no example of the implementation of these functions in the old wiki <https://wiki.openems.de/> or the new documentation. Please give some insight on using this feature!

Hello, I think there is a misunderstanding. In openEMS is no combination of cylindrical and cartesian mesh possible. The MultiGrid is a function to reduce the number of azimuthal lines towards the center of the cylindrical coordinate system (effectively reducing the resolution), as there will appear extremely small mesh elements due to the radius dependency. Without this, the time-step of the FDTD simulation would raise to extreme small values and the simulation time will be huge.
I think the wording in the documentation of MultiGrid is a bit misleading.

ALEEF02, 2024-07-10:
Ah, I see. Thank you.

closed 2024-07-10 by ALEEF02