thliebig/openEMS #152: openEMS Compile issue

hmohajeri, 2024-10-26:
I am having problem compiling with the new CSXCAD and get teh following errors. Is there a fix for this?

 24%] Building CXX object CMakeFiles/openEMS.dir/FDTD/operator_sse_compressed.cpp.o
[ 26%] Building CXX object CMakeFiles/openEMS.dir/FDTD/engine_sse_compressed.cpp.o
[ 27%] Building CXX object CMakeFiles/openEMS.dir/FDTD/operator_multithread.cpp.o
/home/hessam/openEMS-project/openEMS/FDTD/operator.cpp: In member function ‘virtual bool Operator::IsLEparRC(const CSPropLumpedElement*)’:
/home/hm/openEMS-project/openEMS/FDTD/operator.cpp:1716:30: error: ‘LEtype’ is not a member of ‘CSPropLumpedElement’
 1716 |         CSPropLumpedElement::LEtype lumpedType = p_prop->GetLEtype();
      |                              ^~~~~~
/home/hm/openEMS-project/openEMS/FDTD/operator.cpp:1720:30: error: ‘lumpedType’ was not declared in this scope
 1720 |         bool IsParallelRC = (lumpedType == CSPropLumpedElement::PARALLEL) && !(L > 0.0);
      |                              ^~~~~~~~~~
/home/hm/openEMS-project/openEMS/FDTD/operator.cpp:1720:65: error: ‘PARALLEL’ is not a member of ‘CSPropLumpedElement’
 1720 |         bool IsParallelRC = (lumpedType == CSPropLumpedElement::PARALLEL) && !(L > 0.0);
      |                                                                 ^~~~~~~~

thliebig, 2024-11-10:
It compiles fine for me, please try the current release or current git version?
Make sure openEMS and CSXCAD are in sync... e.g. by using the parent project openEMS-Project

hmohajeri, 2024-11-11:
Thanks for the response. I tried again using the latest git depository but I still get the same error.

/home/hessam/openEMS-Project/openEMS/FDTD/operator.cpp: In member function ‘virtual bool Operator::IsLEparRC(const CSPropLumpedElement*)’:
/home/hessam/openEMS-Project/openEMS/FDTD/operator.cpp:1716:30: error: ‘LEtype’ is not a member of ‘CSPropLumpedElement’
 1716 |         CSPropLumpedElement::LEtype lumpedType = p_prop->GetLEtype();
      |                              ^~~~~~
/home/hessam/openEMS-Project/openEMS/FDTD/operator.cpp:1720:30: error: ‘lumpedType’ was not declared in this scope
 1720 |         bool IsParallelRC = (lumpedType == CSPropLumpedElement::PARALLEL) && !(L > 0.0);
      |                              ^~~~~~~~~~
/home/hessam/openEMS-Project/openEMS/FDTD/operator.cpp:1720:65: error: ‘PARALLEL’ is not a member of ‘CSPropLumpedElement’
 1720 |         bool IsParallelRC = (lumpedType == CSPropLumpedElement::PARALLEL) && !(L > 0.0);
      |                                                                 ^~~~~~~~
make[2]: *** [CMakeFiles/openEMS.dir/build.make:244: CMakeFiles/openEMS.dir/FDTD/operator.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:176: CMakeFiles/openEMS.dir/all] Error 2
      |                              ^~~~~~

biergaizi, 2024-11-23:
It looks like a search path problem. You have an old copy of CSXCAD installed somewhere before Lumped Circuit was added, but you're trying to compile a new copy of openEMS with Lumped Circuit.

Clean your system up by:

1. Delete all Git repositories, recursively.
2. Delete all CSXCAD and openEMS headers from your system, including `/usr/local`, `~/.local` and other paths.
3. Reclone the full openEMS repo, recursively. Reinstall fparser, CSXCAD, and openEMS, in that order.
4. If it still doesn't work, create a new account and try.

hmohajeri, 2024-11-25:
Thanks biergaizi, indeed that was the issue and old libraries were hanging around in the system. Now it compiles with no issue and all is good.

all the best,

biergaizi, 2024-12-23:
Great to hear that it's now working for you. This issue can now be closed.