thliebig/openEMS #155: Why does `openEMS.AddLumpedPort` asks for `port_nr`?

SengerM, 2024-11-09:
I understand `port_nr` is an incremental number to be used e.g. like this:
```python
import CSXCAD
from openEMS import openEMS

CSX = CSXCAD.ContinuousStructure()
simulation = openEMS()
simulation.SetCSX(CSX)

# Port in:
simulation.AddLumpedPort(
	port_nr = 1,
	R = 50, # Ohm
	start = port_in_start,
	stop = port_in_start + port_size,
	p_dir = 'z',
	excite = 5555, # V m⁻¹
)
# Port out:
simulation.AddLumpedPort(
	port_nr = 2,
	R = 50, # Ohm
	start = port_out_start,
	stop = port_out_start + port_size,
	p_dir = 'z',
	excite = 0, # V m⁻¹
)

```
The user has to call it with a different `port_nr` each time.

Is there any reason for this? I am asking because I confused and mistakenly wrote:
```python
# Port in:
simulation.AddLumpedPort(
	port_nr = 1,
	R = 50, # Ohm
	start = port_in_start,
	stop = port_in_start + port_size,
	p_dir = 'z',
	excite = 5555, # V m⁻¹
)
# Port out:
simulation.AddLumpedPort(
	port_nr = 2,
	R = 50, # Ohm
	start = port_out_start,
	stop = port_out_start + port_size,
	p_dir = 'z',
	excite = 0, # V m⁻¹
)
```
and it failed in the CSV files.

thliebig, 2024-11-09:
I'm not sure the port number cannot repeat? But if it does occur multiple times it means that the voltages and current for this port will be combined into one value and into one file, e.g. ut1
You need separate numbers to record all port values separate.

thliebig, 2024-11-09:
And I guess this would have been more suitable as a question for an openEMS-Project discussion  instead of an issue?

SengerM, 2024-11-09:
Just to be clear, I am not interested in repeating the port number. This was just a mistake I made because I copy-pasted the code and forgot to change it. And I noticed the problem only when trying to open the CSV and was corrupted. I would have preferred 1) that the program raises some error or 2) not even have to care about the port number as a user.

What is the purpose of `port_nr`? What would I use it for?

biergaizi, 2024-11-10:
> What is the purpose of port_nr? What would I use it for?

I believe that theoretically, one can specify the port number just once during port creation, and later in post-processing, one can generate S-parameters such as S11, S21, S31, S41, etc. by enumerating all ports and extract the `port_nr`, although it has not been used in the examples.

SengerM, 2024-11-10:
> > What is the purpose of port_nr? What would I use it for?
> 
> I believe that theoretically, one can specify the port number just once during port creation, and later in post-processing, one can generate S-parameters such as S11, S21, S31, S41, etc. by enumerating all ports and extract the `port_nr` property from them, although it has not been used in the examples.

Ok, in that case `port_nr` is just a unique label for each port so
- It must not be repeated
- It could be a string, making it more clear, especially for complex systems with many ports

thliebig, 2024-11-10:
No the number can repeat, the ports will then "work as one". This can be desired.
No the ports must be numbers, as s1_1 and s2_1 also implies... 
It is normal in RF design that your ports are numbered, like on a network analyzer, again such that you know what s1_1 actually means...

SengerM, 2024-11-10:
In my case I (unintentionally) repeated the numbers and the result seems to be buggy, maybe it is because of another reason, I thought it was because of this, my apologies. 

When both ports had `port_nr=1`, the CSV file for the current, `port_it_1`, was produced like this:
```
% time-domain current integration by openEMS v0.0.36-20-gceb80b5 @Sat Nov  9 11:25:16 2024
% start-coordinates: (0.0205294,-0.00136652,0.000821429) m -> [195,98,108]
% stop-coordinates: (0.0220226,0.00117195,0.000821429) m -> [201,115,108]
% t/s	current77.03055403865e-14	022.10916621159e-13	0
3.51527701932e-13	0
4.92138782705e-13	0
6.32749863478e-13	0
...
8.37971735866e-10	-2.53300476074
8.38112346947e-10	-2.5328564643988.38252958028e-10	-2.53272128108.8.38393569109e-10	-2.5325908668.38.38534180189e-10	-2.532457598.38.3867479127e-10	-2.532312158.388.38815402351e-10	-2.53214718.3898.38956013432e-10	-2.5319558.39098.39096624512e-10	-2.531738.392378.39237235593e-10	-2.53148.3937788.39377846674e-10	-2.5318.39518458.39518457755e-10	-2.538.39659068.39659068836e-10	-2.538.397996798.39799679916e-10	-2.58.3994029098.39940290997e-10	-28.400809020788.40080902078e-10	-8.40221513159e8.40221513159e-10	8.40362124239e-8.40362124239e-108.4050273532e-108.4050273532e-18.40643346401e-108.40643346401e-8.40783957482e-108.40783957482e-8.40924568563e-10	8.40924568563e8.41065179643e-10	28.410651796438.41205790724e-10	2.8.41205790728.41346401805e-10	2.58.4134640188.41487012886e-10	2.58.4148701288.41627623966e-10	2.548.416276238.41768235047e-10	2.5498.41768238.41908846128e-10	2.55058.4190888.42049457209e-10	2.551728.420498.42190068289e-10	2.5529128.42198.4233067937e-10	2.554109818.428.42471290451e-10	2.555313828.428.42611901532e-10	2.5565245158.48.42752512613e-10	2.5577421188.48.42893123693e-10	2.55896830558.8.43033734774e-10	2.5602068901188.43174345855e-10	2.56146144867
8.43314956936e-10	2.56273531914
8.43455568016e-10	2.56403160095
8.43596179097e-10	2.56535196304
...
```
As seen, there seems to be a mistake in line 4 `% t/s	current77.03055403865e-14	022.10916621159e-13	0`  and in line 5963 it kind of merged multiple lines together.

The total number of lines in each file is also quite different:
- `port_ut_1`: 40148 lines
- `port_it_1`: 40991 lines

After changing `port_nr` to `1` and `2` (as I originally was intending to do) the number of lines in all the CSV files (`port_ut_1`, `port_it_1`,  `port_ut_2` and `port_it_2`) was the same, and the CSVs were not corrupted.

thliebig, 2024-11-10:
I will have a look at it, that's indeed not what is supposed to happen

thliebig, 2026-09-17:
python now also warns about duplicate port numbers. At some point I want to revisit this, but I will close this, as this is now more a feature wish than an issue

closed 2026-09-17 by thliebig