thliebig/openEMS #165: Python package (e.g. NumPy) version requirements are too strict for Debian, numpy.core.multiarray failed to import

biergaizi, 2024-12-09:
The CI/CD system has caught the first compatibility problem before it's even deployed in my test repo.

On Debian 11, running openEMS via Python produces the following error:

    ImportError: numpy.core.multiarray failed to import

This is a binary compatibility problem that should've never happened. But it did. Why? Because setup.py requires NumPy 1.26, but Debian only has NumPy 1.24. So Python finds no system package can satisfy this requirement and it attempts to download that from PyPI instead. But now it downloads the incompatible NumPy v2 package, creating the binary compatibility problem. Same problems exist for h5py and cython as well. Python attempts to download them from PyPI instead of using the system package, although there's no compatibility problem *per se*, at least, not yet.

These version requirements were introduced by @oberstet in commit 5bc31d0255839ec9fcbc21f3180df2c8230c6cc7 and e7620dcd72e7913d8f34822a864baee7c68dce08, but *no* justification was given about why these minimum versions were necessary. If I recall correctly, openEMS used to be compatible even with very ancient Python versions.

Thus, I suggest removing these version requirements entirely. It's a bit sloppy but it's easier than trying to test ancient systems and figuring out the exact minimum versions. If we insist on adding version requirement, it should be at least compatible with Debian's `oldoldstable`.

oberstet, 2024-12-09:
first of all, I'm sorry you've run into issues with that!

but: lacking a dependency policy (doc guidance) and lacking automated CD (defacto guidance) in openEMS made me follow what I usually do:

depend on latest upstream stable

in this case, the commit was Dec 16, 2023, and the latest upstream stable was exactly 1.26.2 (12. Nov. 2023) _at commit time_:

https://pypi.org/project/numpy/#history

> but no justification was given about why these minimum versions were necessary

security - MUST ALWAYS use latest upstream stable (at minimum, otherwise, do you want to follow all upstreams in such detail and backport or only upgrade when necessary functionally?)

> If we insist on adding version requirement, it should be at least compatible with Debian's oldoldstable.

I disagree, the version should have had an upper bound (<2). I missed that there was a major upstream upgrade in the cooking .. the 2.0.0 was release in June 2024

Debian (and other distros) repackaging of Python packages is wrong altogether, upstream Python packages should be installed into dedicated Python virtualenvs, and not into a base OS anyways

But again, I have strong opinions, I'm aware, and I have been fighting such crap myself in my own OSS packages (e.g. Crossbar.io which has almost 200 Python dependencies), and up to a point where Debian maintainers of the Crossbar.io deb repackage gave up;) Good for me, no more pointless discussions.

biergaizi, 2024-12-09:
> security - MUST ALWAYS use latest upstream stable (at minimum, otherwise, do you want to follow all upstreams in such detail and backport or only upgrade when necessary functionally?)

First, it's my understanding that the unwritten project policy of openEMS attempts to maintain compatibility when it's practical, and rarely drops anything explicitly. Qt4, Qt5, Qt6 are currently maintained. C++11 compliance is still maintained (there's a regression that I'm going to fix). VTK7, VTK8, VTK9 compatibility are maintained, etc. So openEMS basically has a CentOS-like policy of 10-year backward compatibility. If we want to change the policy systematically - which may not be an unreasonable decision, it should start from the C++ codebase first, not from the 3rd-tier Python module. It's pathological if the main C++ executable runs but the Python extension does not, and claiming it's for security.

Second, speaking of security: If the end-user insists on running an old system, and if the base system has already stopped receiving any security update, doing so at the application language level does very little. The first root of trust is the kernel, the next root of trust is the libc, and only then we have programming-language level root of trust.

Furthermore, for applications that handle network or untrusted data, it may be practically extremely important to implement security solely from application level since it may be deployed on all kinds of legacy systems. But openEMS 

> Debian (and other distros) repackaging of Python packages is wrong altogether, upstream Python packages should be installed into dedicated Python virtualenvs, and not into a base OS anyways

Python replacing Debian (and other distros) is wrong altogether. Downstream OS packages should be installed into the  base system controlled by the downstream.

In the ideal world, there's no need for language-specific package managers if distros can manage packages in the way nixOS or GNU Guix does and maintain 100% freshness. But of course it's impractical, so I don't advocate for the elimination of language-specific packages, but they should not be the entire replacement of another, both should be allowed to work if practical. The ultimate choice is up to the end-user. If one wants to maintain a system with fully up-to-date upstream package, one is free to use virtualenv as a personal decision. But the upstream project should not break other use cases - especially the use case supported by the project since its inception.

thliebig, 2024-12-09:
> From the history of the project, I think its unwritten dependency policy is pretty clear - never drop any support until the last moment.

100% correct and I agree with everything you said @biergaizi We should maybe create a policy file stating that old dependencies should be supported as long as it is still practical to do so? Something like that?
My goal with openEMS is to be as inclusive as I possibly and with reasonable effort can be. That can mean supporting ancient old Linux if it is possible without (much) extra effort. And as @biergaizi said, this does **not** pose any security issue to anyone building and using openEMS on the latest and newest OS available.

Do you want to create a PR to remove this unnecessary min. version checks? Otherwise I will have a look at it later today.

oberstet, 2024-12-09:
> We should maybe create a policy file stating that 

yep, that would be good - whatever it says - and fwiw, I would have tried to comply with such a document when contributing of course

> And as @biergaizi said, this does not pose any security issue to anyone building and using openEMS on the latest and newest OS available.

my view: yes it does, the risk comes from the dependency chain, and an end-user / developer must then manage not only direct, but also all indirect dependencies himself, the *transitive closure of the whole SBOM* down to the C run-time or further.  if that is the case and expected, it should be stated somewhere for the user to be aware at least, IMO, but anyways, all said, nothing to add or contribute for me

thliebig, 2024-12-09:
I'm not sure I'm able to understand. Lets assume user A downloads and builds the latest openEMS against an ancient version of  numpy on a very old Debian. And lets assume user B does the same against a new numpy on the latest Arch Linux. How does not preventing user A from building the latest openEMS help user B to be more secure???

biergaizi, 2024-12-09:
> I'm not sure I'm able to understand. Lets assume user A downloads and builds the latest openEMS against an ancient version of numpy on a very old Debian. And lets assume user B does the same against [...]

I think I understanding it to a certain extent. The recent industrial trend of containerization is to bypass *the entirety of* the user's base system completely. So even if the base system is ancient, from an application developer's point of view, they still have full control and they're effectively delivering a small `chroot` environment with all the latest libraries. And by dropping support of old systems, you effectively force all of these users to use the official container image with newer libraries instead, so in this sense it's more "secure".

The opinion of doing so is pretty polarized, but I think it's a legitimate practice for (1) server/DevOps automation, and (2) installing applications when the base system is too impractical to change (i.e. too new or too old). So in this sense, I'm not fundamentally against the idea of distributing a Docker image of openEMS, *as an optional choice for end users*. I'm just not a fan of *forcing all users* to use the container image.