thliebig/openEMS #169: MPI build broken: error: ‘parseCommandLineArgument’ is not a member of ‘openEMS’

miek, 2024-12-29:
In trying to build the latest version, I found that it errors out with:
```
/home/mike/repos/openEMS-Project/openEMS/FDTD/openems_fdtd_mpi.cpp: In member function ‘virtual bool openEMS_FDTD_MPI::parseCommandLineArgument(const char*)’:
/home/mike/repos/openEMS-Project/openEMS/FDTD/openems_fdtd_mpi.cpp:118:29: error: ‘parseCommandLineArgument’ is not a member of ‘openEMS’
  118 |         bool ret = openEMS::parseCommandLineArgument( argv );
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~
```
I guess there may have been a missing change in #140 that only shows up if you have MPI enabled.

I was running `./update_openEMS.sh ~/opt/openEMS --with-hyp2mat --with-CTB --with-MPI --python`. I don't actually need MPI, so I'll carry on without it enabled, but just thought I'd let you know.

biergaizi, 2025-10-22:
Noted.

biergaizi, 2025-10-22:
I think it's objectively useful if it allows one to run simulations larger than a single PC's memory size, the official openEMS MRI paper clearly showed that. But yeah, its workflow is undocumented, so few if any researcher is using it (the fact that modern servers have TiB of memory further reduced the needs). Still, since MPI was featured in the original research paper, I guess it's to better to keep the MPI engine rather than removing it outright. Even if it's to be deleted later, perhaps we should release a "last-supported version" with working MPI. 

I do have some ideas on improving the MPI engine. The current implementation is not the most efficient one possible, for example there's a trivial field communication optimization which allows one communication pass per one timestep, rather than half timesteps. More sophisticated optimizations improve performance further, so that can increase MPI's usefulness. But yes, given that the single-node optimization is still unfinished, any MPI work is on a very-long-term TODO list.

thliebig, 2026-09-17:
MPI has been removed. Maybe some day a better version can be readded. Should not be hard for a modern LLM. The old implementation was just completely broken for many many years already.

closed 2026-09-17 by thliebig