thliebig/openEMS #172: Tutorial Simple_Patch_Antenna shows error in plotting S11

dkozel, 2025-01-12:
Hello. I'm just getting started and am running Octave 8.4.0 (I will retest with the latest 9.3.0 tomorrow) and OpenEMS 0.0.36 on Windows 11. I have just tried the Simple Patch Antenna tutorial example and it runs successfully other than one error in the S11 plot.

```
150: h = plot(s11);
error: vertical dimensions mismatch (1x3 vs 1x1)
error: called from
    legend>legend_autoupdate_cb at line 638 column 5
    __plt__>__plt2vv__ at line 502 column 10
    __plt__>__plt2__ at line 248 column 14
    __plt__>__plt1__ at line 207 column 10
    __plt__ at line 121 column 16
    plot at line 235 column 10
    plotRefl at line 150 column 3
    Simple_Patch_Antenna at line 123 column 1
```

https://github.com/thliebig/openEMS/blob/078483d169b178298251398b81763b6f3679bd1b/matlab/Tutorials/Simple_Patch_Antenna.m#L123

Can anyone reproduce this? It doesn't look as if there have been any changes to the example or the plotRefl code in the last year. I'll look into the code more, but wanted to get a report up.

gammaxy1, 2025-01-30:
I reproduced the problem in Octave 6.4.0. It seems related to calling legend() before plot(s11) in plotRefl.m. I swapped the order of the two function calls and it seems to have fixed it.

Another fix is to pass the plot handles to legend() as a column vector instead of a row vector, but I don't know enough about matlab/octave to know if this is expected behavior.

Here's both fixes:
```
legend([handle1; handle2; handle3], {[llegend, " MHz"], ...
                                     [ulegend, " MHz"], ...
                                     [num2str(20*log10(abs(s11(minind))), "%4.0f"), ...
                                    "dB @ ", num2str(port.f(minind)/1e6, ffmt), " MHz"]});

h = plot(s11);
```

thliebig, 2026-09-17:
Should be fixed

closed 2026-09-17 by thliebig

dkozel, 2026-09-27:
Thanks! OpenEMS is a very useful project, thank you for your time and expertise working on it!