thliebig/openEMS #180: Installation issue with homebrew in MacOS - cmake version problem

scubaed-coder, 2025-04-18:
I remember seeing a similar but closed issue when looking around, but it seems to be in a different openEMS repository?

I am out of my depth here trying to figure out the intricacies of home-brew, installing from local repositories and the like, so I am not able to provide much more detail beyond the reported problem itself. I tried editing the offending file locally and the brew formula, but I couldn't get brew to use it properly. I reinstalled brew from scratch (it was an old install that had been following me around through three machines) and tried again with the same results.

Platform:  MacStudio, M2 Max, Sequoia 15.3.2 (24D81)

On using: brew install --HEAD thliebig/openems/openems  I get the following error message:

```
CMake Error at CMakeLists.txt:7 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.

-- Configuring incomplete, errors occurred!

```

If i bypass the issue using brew install --debug --HEAD thliebig/openems/openems  and running the offending line adding the suggested version switch, it breaks later in the make process.

```
ailed executing: cmake . -DCMAKE_INSTALL_PREFIX=/opt/homebrew/Cellar/openems/HEAD-9ba6e1b -DCMAKE_INSTALL_LIBDIR=lib -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_PROJECT_TOP_LEVEL_INCLUDES=/opt/homebrew/Library/Homebrew/cmake/trap_fetchcontent_provider.cmake -Wno-dev -DBUILD_TESTING=OFF -DCMAKE_OSX_SYSROOT=/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk
1. raise
2. ignore
3. backtrace
4. irb
5. shell
Choose an action: 5

openems-20250418-65188-29z6h5 % cmake . -DCMAKE_INSTALL_PREFIX=/opt/homebrew/Cellar/openems/HEAD-9ba6e1b -DCMAKE_INSTALL_LIBDIR=lib -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_PROJECT_TOP_LEVEL_INCLUDES=/opt/homebrew/Library/Homebrew/cmake/trap_fetchcontent_provider.cmake -Wno-dev -DBUILD_TESTING=OFF -DCMAKE_OSX_SYSROOT=/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk -DCMAKE_POLICY_VERSION_MINIMUM=3.5

-- Using CMAKE_INSTALL_PREFIX: /opt/homebrew/Cellar/openems/HEAD-9ba6e1b
-- Build AppCSXCAD: YES
-- with MPI: 
-- Configuring done (0.1s)
-- Generating done (0.0s)
-- Build files have been written to: /private/tmp/openems-20250418-65188-29z6h5

openems-20250418-65188-29z6h5 % exit
1. raise
2. ignore
3. backtrace
4. irb
5. shell
Choose an action: 2
==> make
Last 15 lines from /Users/ebrown/Library/Logs/Homebrew/openems/02.make:
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:7 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.

-- Configuring incomplete, errors occurred!
```

Any ideas/fixes/patches?

scubaed-coder, 2025-04-18:
Oh, I also tried editing the offending makefile, with similar results.

```
BuildError: Failed executing: cmake . -DCMAKE_INSTALL_PREFIX=/opt/homebrew/Cellar/openems/HEAD-9ba6e1b -DCMAKE_INSTALL_LIBDIR=lib -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_PROJECT_TOP_LEVEL_INCLUDES=/opt/homebrew/Library/Homebrew/cmake/trap_fetchcontent_provider.cmake -Wno-dev -DBUILD_TESTING=OFF -DCMAKE_OSX_SYSROOT=/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk
1. raise
2. ignore
3. backtrace
4. irb
5. shell
Choose an action: 5
 openems-20250418-69394-khe5y % vi CMakeLists.txt [change CMakeLists.txt:7 to cmake_minimum_required(VERSION 3.5)]
openems-20250418-69394-khe5y % 

```

ammaratef45, 2025-05-06:
I have the same issue

scubaedgar, 2025-05-09:
I am not sure if any of this helps, but I haven't seen any activity so far:

From [StackOverflow:](https://stackoverflow.com/questions/79534856/cannot-build-cmake-project-because-compatibility-with-cmake-3-5-has-been-remo)

> Is the error saying that my CMake version is too new to build this project? Do I need to install an older version of CMake?
> Yes, this is the exact meaning of the error message: CMake 4.0.0 no longer supports features deprecated in CMake 3.5 or earlier.
> 
> If a project actually uses such deprecated features, then you need to build the project with CMake 3.31 or older.
> 
> If a project doesn't use such deprecated features, but is written with
> 
> cmake_minimum_required(VERSION 3.4) # or lower CMake version
> then you could redefine a project's VERSION specification via setting CMAKE_POLICY_VERSION_MINIMUM parameter.

From [phpBB:](https://forum.freecad.org/viewtopic.php?style=1&t=95162)

> I've found that if I change the first line of coin-config.cmake to specify CMake 3.5 instead of 2.8.12, this becomes a warning, and if I change it to 3.10 the message goes away entirely, but this should not be the proper work-around. The suggestion in the error to add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 seems equally suspicious to me insofar as it seems to be just wallpapering over some compatibility problem.

Apparently related bug reports:

- https://github.com/bbc/homebrew-audiowaveform/issues/9
- 
- https://github.com/AOMediaCodec/libavif/issues/2724
- 
- https://github.com/matrix-nio/matrix-nio/issues/541
- 
- https://issues.apache.org/jira/browse/THRIFT-5860

Gadgeteering, 2025-05-10:
If you edit the CMakeLists.txt files and search for cmake_minimum_required() and change it to cmake_minimum_required(VERSION 4.0) it complies

MayeulC, 2025-05-14:
Hi, I just encountered this compatibility issue when building https://github.com/matrix-nio/matrix-nio/, and found this mention. Just letting you know that you can define the `CMAKE_POLICY_VERSION_MINIMUM=3.5` environment variable during compilation to make things easier, as a user. Of course, it still needs to be fixed in upstream cmakefiles.

biergaizi, 2025-06-21:
Fixed in https://github.com/thliebig/openEMS-Project/pull/340

closed 2025-10-28 by thliebig